eclipse-threadx/levelx #3: _lx_nor_flash_logical_sector_find takes too much time(about 140ms) when lx_nor_flash_total_blocks = 8192

nstcl, 2020-08-15:
I use W25Q256 QSPI flash. Its block size is 4K bytes, chip size is 32M bytes, total blocks is 8192. When I write data to file, _lx_nor_flash_logical_sector_find  tasks about 140 ms to read all blocks sometimes. So the data writing takes too much time.

goldscott, 2020-08-17:
Hi @nstcl - are you using FileX or just LevelX?

There is a logical sector cache in LevelX, this can be increased for better performance (see LevelX User Guide).

nstcl, 2020-08-18:
I use FileX with LevelX.

nstcl, 2020-08-18:
I use FileX with LevelX. I change flash area used for file system from 32M bytes to 16M bytes, total blocks = 4096.  
#define LX_NOR_SECTOR_MAPPING_CACHE_SIZE            128 
#define LX_NOR_EXTENDED_CACHE_SIZE                  64
I create a new file and write 4K bytes data to it. I add debug info in function _lx_nor_flash_logical_sector_find.
[_lx_nor_flash_logical_sector_find:219]:[tick:18491] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:18552] total_blocks=-1
[_lx_nor_flash_logical_sector_find:106]:[tick:18558]
[_lx_nor_flash_logical_sector_find:219]:[tick:18558] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:18619] total_blocks=-1
[_lx_nor_flash_logical_sector_find:106]:[tick:18627]
[_lx_nor_flash_logical_sector_find:219]:[tick:18627] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:18688] total_blocks=-1
[_lx_nor_flash_logical_sector_find:106]:[tick:18694]
[_lx_nor_flash_logical_sector_find:219]:[tick:18694] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:18755] total_blocks=-1
[_lx_nor_flash_logical_sector_find:106]:[tick:18761]
[_lx_nor_flash_logical_sector_find:219]:[tick:18761] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:18822] total_blocks=-1
[_lx_nor_flash_logical_sector_find:106]:[tick:18828]
[_lx_nor_flash_logical_sector_find:219]:[tick:18828] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:18889] total_blocks=-1
[_lx_nor_flash_logical_sector_find:106]:[tick:18895]
[_lx_nor_flash_logical_sector_find:219]:[tick:18895] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:18956] total_blocks=-1
[_lx_nor_flash_logical_sector_find:106]:[tick:18962]
[_lx_nor_flash_logical_sector_find:219]:[tick:18962] total_blocks=4096
[_lx_nor_flash_logical_sector_find:477]:[tick:19023] total_blocks=-1
As I can see, when each sector is writing, _lx_nor_flash_logical_sector_find read 4096 flash blocks to find the logical sector is mapped or not. This step takes about 60ms. Is there any other options I can make to improve performance.
Thanks

goldscott, 2020-08-18:
Using FileX, we generally suggest that the FileX format leaves one block worth of sectors free, i.e., the format is smaller than the flash size by one flash block. Are you doing this?

For NOR flash, there is a #define to directly read the flash rather than going through the driver for reading (see LevelX User Guide).

nstcl, 2020-08-18:
#define EXT_FLASH_FS_LENGTH              0x01000000
#define EXT_FLASH_SECTOR_SIZE                0x1000

goldscott, 2020-08-18:
What does your cache setup look like? Are you using the lx_nor_flash_extended_cache_enable API?  If not, try to experiment with it.  Another cache tuning parameter is the LX_NOR_SECTOR_MAPPING_CACHE_SIZE define. The default for this is 16 and  must be a power of 2. Perhaps increasing this to 32 or 64 or 128 would be useful to experiment with.

closed 2020-08-27 by goldscott