eclipse-threadx/levelx #4: Assert is happen in lx_nand_flash_open()  API

wxp821217, 2020-08-15:
Dear LevelX,
   
    I have tested demo with SPINAND as your nand flash demo. 
    Some  Assert problem is happened very time.

`
[image]
`

1. lx_nand_flash_open is called by fx_media_format firstly.
2. lx_nand_flash_open is called by fx_media_open again.

As you seen, The follow is in problem. 
` LX_PARAMETER_NOT_USED(name);

    /* Clear the NAND flash control block.  */

    TX_MEMSET(nand_flash, 0, sizeof(LX_NAND_FLASH));

    

    /* Call the flash driver's initialization function.  */

    (nand_driver_initialize)(nand_flash);`

`    if (_lx_nand_flash_opened_count)

    {

        /* List is not empty - other NAND flashes are open.  */

        /* Pickup tail pointer.  */

        tail_ptr =  _lx_nand_flash_opened_ptr -> lx_nand_flash_open_previous;

        /* Place the new NAND flash control block in the list.  */

        _lx_nand_flash_opened_ptr -> lx_nand_flash_open_previous =  nand_flash;

        tail_ptr -> lx_nand_flash_open_next =                       nand_flash;

        /* Setup this NAND flash's opened links.  */

        nand_flash -> lx_nand_flash_open_previous =  tail_ptr;

        nand_flash -> lx_nand_flash_open_next =      _lx_nand_flash_opened_ptr;   

    }`

   Because of memst of nand flash, some assert is happened.

Best Regards
Joshu Wang

goldscott, 2020-08-17:
Hi @wxp821217 - thanks for letting us know about this issue. We're looking into it. I'll get back to you soon.

goldscott, 2020-08-18:
Hi @wxp821217 - you are correct, `fx_media_format` makes a driver request `FX_DRIVER_INIT`, which calls `_lx_nand_flash_open`. At the end of `fx_media_format`, a `FX_DRIVER_UNINIT` driver request is made, which calls `_lx_nand_flash_close`. Later, when you call `fx_media_open`, another driver request `FX_DRIVER_INIT` is made and `_lx_nand_flash_open` is called again. This is how FileX and LevelX are supposed to work. Are you seeing problems?

wxp821217, 2020-08-20:
Dear goltscott,
   
     Some Abort is happened very time when I am doing some try with Demo_filex_nand_flash.

    With logs, Root cause is likely using memset() unreasonably.
   
1.  _lx_nand_flash_open is called firstly.
   * memset(nandflash)
   * And open operation chain as follows.

`
    else

    {

        /* The opened NAND flash list is empty.  Add the NAND flash to empty list.  */

        _lx_nand_flash_opened_ptr =                 nand_flash;

        nand_flash -> lx_nand_flash_open_next =      nand_flash;

        nand_flash -> lx_nand_flash_open_previous =  nand_flash;

    }`

2. _lx_nand_flash_open is called firstly.
   * memset(nandflash)
   * And open operation chain as follows

`
    /* Place the NAND flash control block on the list of opened NAND flashes.  First,

       check for an empty list.  */

    if (_lx_nand_flash_opened_count)

    {

        /* List is not empty - other NAND flashes are open.  */

        /* Pickup tail pointer.  */

        tail_ptr =  _lx_nand_flash_opened_ptr -> lx_nand_flash_open_previous;

        /* Place the new NAND flash control block in the list.  */

        _lx_nand_flash_opened_ptr -> lx_nand_flash_open_previous =  nand_flash;

        tail_ptr -> lx_nand_flash_open_next =                       nand_flash;

        /* Setup this NAND flash's opened links.  */

        nand_flash -> lx_nand_flash_open_previous =  tail_ptr;

        nand_flash -> lx_nand_flash_open_next =      _lx_nand_flash_opened_ptr;   

    }`

But Here nand_flash -> lx_nand_flash_open_previous=NULL,(Because of memset at fistly), So the   nand_flash -> lx_nand_flash_open_previous =  tail_ptr  is Illegal.

  I will double check it again..

Best Regards
Joshua Wang

goldscott, 2020-08-20:
@wxp821217 - in the demo_filex_nand_flash.c example, when `fx_media_open` is called, `_lx_nand_flash_opened_count` is 0. So the else statement is executed.

wxp821217, 2020-08-27:
Dear goldscott,

    Pls Close it. Thanks.

Best Regards
Joshua Wang

goldscott, 2020-08-27:
Hi @wxp821217 - sounds like you need to step into you driver and figure out why the FX_DRIVER_BOOT_WRITE request is failing.

goldscott, 2020-09-01:
closing - feel free to open again.

closed 2020-09-01 by goldscott