eclipse-threadx/levelx #5: levelX doesnot work-well

wxp821217, 2020-08-15:
Dear levelx,

    I have done much work with levelx In my platform.
   cpu : Cortex R5
   Flahs: Winbond SPINAND.
   
   But it is donot work-well.
  As Power On, The spare area with nand is all 0xFFFFFFFF, So all important fucntion is not called , And the result is failed.

` /* Determine if the page is free.   */

                    if (extra_info.lx_nand_page_extra_info_logical_sector == LX_NAND_PAGE_FREE)

                    {

                    

                        /* The page is free and we know that the pages are allocated sequentially

                           so there is no need to go further.  */

                    

                        /* Increment the free pages counter.  */

                        free_pages =  free_pages + (nand_flash -> lx_nand_flash_pages_per_block - page);

                        

                        /* Verify that the same position in the mapped list from page 0 is also free.  */

                        if (page_word_ptr[page] != LX_NAND_PAGE_FREE)

                        {

                            

                            /* Increment the invalid mapping counter.  */

                            nand_flash -> lx_nand_flash_diagnostic_mapping_invalid++;

                            

                            /* Call system error handler.  */

                            _lx_nand_flash_system_error(nand_flash, LX_SYSTEM_INVALID_FORMAT, block, page);

                        }

                        

                        /* Determine if the associated page is really free... it could be the case that the 

                           page data area was in the process of being written or just completed when a power

                           interruption occurs before the extra bytes are setup with the logical sector.  */

                        /* Read verify the page is erased.  */

                        status =  _lx_nand_flash_driver_page_erased_verify(nand_flash, block, page);

                        /* Check for an error from flash driver.   */

                        if (status == LX_ERROR)

                        {

                            /* Increment the page data not free diagnostic counter.  */

                            nand_flash -> lx_nand_flash_diagnostic_page_data_not_free++;

                            

                            /* Decrement the free page counter and increment the obsolete page counter.  */

                            free_pages--;

                            obsolete_pages++;

                            

                            /* Now setup the extra info to show this page is obsolete.  */

                            extra_info.lx_nand_page_extra_info_logical_sector =  0;

                            status =  _lx_nand_flash_driver_extra_bytes_set(nand_flash, block, page, (UCHAR *) &extra_info, sizeof(extra_info));

                            /* Check for an error from flash driver.   */

                            if (status)

                            {

        

                                /* Call system error handler.  */

                                _lx_nand_flash_system_error(nand_flash, status, block, page);

                                /* Return an error.  */

                                return(LX_ERROR);

                            }

                        }

                           

                        /* At this point we can break out of the page traversal loop, since nothing else can exist after this page.  */

                        break;

                    }`

  Some question is listed here.
1. How to initlized the spare zone of per page of nand. 
 If it is all 0XFFFFFFFF, it is in probem.

 Thanks.
Best Regards

Joshua Wang

goldscott, 2020-08-17:
Hi @wxp821217 - Can you tell me where in the code you pasted that is failing? LX_NAND_PAGE_FREE is defined to 0xFFFFFFFF. So this code should all be called.

wxp821217, 2020-08-18:
Dear goldscott,

    lots of logs is added for the issue.
    I will show you briefly.

`                   /* Determine if the page is free.   */

                    if (extra_info.lx_nand_page_extra_info_logical_sector == LX_NAND_PAGE_FREE)

                    {

                    

                        /* The page is free and we know that the pages are allocated sequentially

                           so there is no need to go further.  */

                    

                        /* Increment the free pages counter.  */

                        free_pages =  free_pages + (nand_flash -> lx_nand_flash_pages_per_block - page);

                        

                        /* Verify that the same position in the mapped list from page 0 is also free.  */

                        if (page_word_ptr[page] != LX_NAND_PAGE_FREE)

                        {

                            

                            /* Increment the invalid mapping counter.  */

                            nand_flash -> lx_nand_flash_diagnostic_mapping_invalid++;

                            

                            /* Call system error handler.  */

                            _lx_nand_flash_system_error(nand_flash, LX_SYSTEM_INVALID_FORMAT, block, page);

                        }

                        

                        /* Determine if the associated page is really free... it could be the case that the 

                           page data area was in the process of being written or just completed when a power

                           interruption occurs before the extra bytes are setup with the logical sector.  */

                        /* Read verify the page is erased.  */

                        status =  _lx_nand_flash_driver_page_erased_verify(nand_flash, block, page);

                        /* Check for an error from flash driver.   */

                        if (status == LX_ERROR)

                        {

                            /* Increment the page data not free diagnostic counter.  */

                            nand_flash -> lx_nand_flash_diagnostic_page_data_not_free++;

                            

                            /* Decrement the free page counter and increment the obsolete page counter.  */

                            free_pages--;

                            obsolete_pages++;

                            

                            /* Now setup the extra info to show this page is obsolete.  */

                            extra_info.lx_nand_page_extra_info_logical_sector =  0;

                            status =  _lx_nand_flash_driver_extra_bytes_set(nand_flash, block, page, (UCHAR *) &extra_info, sizeof(extra_info));

                            /* Check for an error from flash driver.   */

                            if (status)

                            {

        

                                /* Call system error handler.  */

                                _lx_nand_flash_system_error(nand_flash, status, block, page);

                                /* Return an error.  */

                                return(LX_ERROR);

                            }

                        }

                           

                        /* At this point we can break out of the page traversal loop, since nothing else can exist after this page.  */

                        break;

                    }`

  It is returned here very time, if Spare area = 0XFFFFFFFFF.
  As you known, lots of important info is lost, such as nand_flash -> lx_nand_flash_mapped_pages, so LEVELX is failed.

  Pls note the follows API.

`UINT  _lx_nand_flash_logical_sector_find(LX_NAND_FLASH *nand_flash, ULONG logical_sector, ULONG superceded_check, ULONG *block, ULONG *page)

{

LX_NAND_PAGE_EXTRA_INFO             extra_info;

UCHAR                               block_status;

ULONG                               *block_word_ptr;

ULONG                               valid_pages;

ULONG                               mapped_pages;

ULONG                               total_blocks;

ULONG                               total_pages;

ULONG                               i, j;

ULONG                               search_start;

ULONG                               max_mapped_sector;

#ifndef LX_NAND_FLASH_DIRECT_MAPPING_CACHE

LX_NAND_SECTOR_MAPPING_CACHE_ENTRY  *sector_mapping_cache_entry_ptr = LX_NULL;

LX_NAND_SECTOR_MAPPING_CACHE_ENTRY  temp_sector_mapping_cache_entry;

#endif

UINT                                status;

    /* Initialize the return parameters.  */

    *block =  (ULONG) 0;

    *page =   (ULONG) 0;

    

    /* Determine if there are any mapped pages.  */

    if (nand_flash -> lx_nand_flash_mapped_pages == 0)

    {

        /* No mapped pages so nothing can be found!.  */

        return(LX_SECTOR_NOT_FOUND);

    }`

  LX_SECTOR_NOT_FOUND is returned very time.. 

 Best Regards
Joshua Wang

goldscott, 2020-08-18:
I'm having a hard time following you. What line of lx_nand_flash_open.c is returning an error? What does your `LX_NAND_PAGE_EXTRA_INFO extra_info` struct look like? Is your driver reading your flash correctly?

wxp821217, 2020-08-20:
Dear Goldscott,
     
      Sorry for my poor English, I need to study your English hardly.
  
   1. There is no error returned in x_nand_flash_open.c API.
    [2020/8/15 18:55:44-670] _lx_nand_flash_open7[ffffffff]
[2020/8/15 18:55:44-689] _lx_nand_flash_dev_extra_bytes_get->[1][1][4]
[2020/8/15 18:55:44-689] SPINAND_Read_Page_SpareData: [0x20800]!!
[2020/8/15 18:55:44-689] _lx_nand_flash_dev_extra_bytes_get2->[ffffffff]
[2020/8/15 18:55:44-689] _lx_nand_flash_dev_page_erased_verify->[1][1]
[2020/8/15 18:55:44-711] SPINAND_Read: [0x20800][0x800]!!
[2020/8/15 18:55:44-712] SPINAND_Read_Page_SpareData: [0x40000]!!
[2020/8/15 18:55:44-712] _lx_nand_flash_dev_block_status_get->[2][ff]
[2020/8/15 18:55:44-712] _lx_nand_flash_driver_read3
[2020/8/15 18:55:44-712] _lx_nand_flash_dev_read->[2][0][65]
[2020/8/15 18:55:44-712] SPINAND_Read: [0x40000][0x104]!!
[2020/8/15 18:55:44-765] _lx_nand_flash_open5[14]
[2020/8/15 18:55:44-765] _lx_nand_flash_open7[ffffffff]
[2020/8/15 18:55:44-765] _lx_nand_flash_dev_extra_bytes_get->[2][1][4]
[2020/8/15 18:55:44-765] SPINAND_Read_Page_SpareData: [0x40800]!!
[2020/8/15 18:55:44-765] _lx_nand_flash_dev_extra_bytes_get2->[ffffffff]
[2020/8/15 18:55:44-765] _lx_nand_flash_dev_page_erased_verify->[2][1]
[2020/8/15 18:55:44-765] SPINAND_Read: [0x40800][0x800]!!
[2020/8/15 18:55:44-765] SPINAND_Read_Page_SpareData: [0x60000]!!
[2020/8/15 18:55:44-766] _lx_nand_flash_dev_block_status_get->[3][ff]
[2020/8/15 18:55:44-766] _lx_nand_flash_driver_read3
[2020/8/15 18:55:44-766] _lx_nand_flash_dev_read->[3][0][65]
[2020/8/15 18:55:44-766] SPINAND_Read: [0x60000][0x104]!!
[2020/8/15 18:55:44-766] _lx_nand_flash_open5[14]
[2020/8/15 18:55:44-766] _lx_nand_flash_open7[ffffffff]
[2020/8/15 18:55:44-784] _lx_nand_flash_dev_extra_bytes_get->[3][1][4]
[2020/8/15 18:55:44-784] SPINAND_Read_Page_SpareData: [0x60800]!!
[2020/8/15 18:55:44-784] _lx_nand_flash_dev_extra_bytes_get2->[ffffffff]
[2020/8/15 18:55:44-784] _lx_nand_flash_dev_page_erased_verify->[3][1]
[2020/8/15 18:55:44-803] SPINAND_Read: [0x60800][0x800]!!
[2020/8/15 18:55:44-803] SPINAND_Read_Page_SpareData: [0x80000]!!
[2020/8/15 18:55:44-803] _lx_nand_flash_dev_block_status_get->[4][ff]
[2020/8/15 18:55:44-803] _lx_nand_flash_driver_read3
[2020/8/15 18:55:44-803] _lx_nand_flash_dev_read->[4][0][65]
[2020/8/15 18:55:44-823] SPINAND_Read: [0x80000][0x104]!!
[2020/8/15 18:55:44-823] _lx_nand_flash_open5[14]
[2020/8/15 18:55:44-823] _lx_nand_flash_open7[ffffffff]
[2020/8/15 18:55:44-823] _lx_nand_flash_dev_extra_bytes_get->[4][1][4]
[2020/8/15 18:55:44-823] SPINAND_Read_Page_SpareData: [0x80800]!!
[2020/8/15 18:55:44-842] _lx_nand_flash_dev_extra_bytes_get2->[ffffffff]
[2020/8/15 18:55:44-842] _lx_nand_flash_dev_page_erased_verify->[4][1]
[2020/8/15 18:55:44-842] SPINAND_Read: [0x80800][0x800]!!
[2020/8/15 18:55:44-842] SPINAND_Read_Page_SpareData: [0xa0000]!!
[2020/8/15 18:55:44-863] _lx_nand_flash_dev_block_status_get->[5][ff]
[2020/8/15 18:55:44-863] _lx_nand_flash_driver_read3
[2020/8/15 18:55:44-863] _lx_nand_flash_dev_read->[5][0][65]
[2020/8/15 18:55:44-863] SPINAND_Read: [0xa0000][0x104]!!
[2020/8/15 18:55:44-863] _lx_nand_flash_open5[13]
[2020/8/15 18:55:44-863] _lx_nand_flash_open7[ffffffff]
[2020/8/15 18:55:44-887] _lx_nand_flash_dev_extra_bytes_get->[5][1][4]
[2020/8/15 18:55:44-887] SPINAND_Read_Page_SpareData: [0xa0800]!!
[2020/8/15 18:55:44-887] _lx_nand_flash_dev_extra_bytes_get2->[ffffffff]
[2020/8/15 18:55:44-887] _lx_nand_flash_dev_page_erased_verify->[5][1]
[2020/8/15 18:55:44-887] SPINAND_Read: [0xa0800][0x800]!!
[2020/8/15 18:55:44-888] SPINAND_Read_Page_SpareData: [0xc0000]!!
[2020/8/15 18:55:44-933] _lx_nand_flash_dev_block_status_get->[6][ff]
[2020/8/15 18:55:44-933] _lx_nand_flash_driver_read3
[2020/8/15 18:55:44-933] _lx_nand_flash_dev_read->[6][0][65]
[2020/8/15 18:55:44-933] SPINAND_Read: [0xc0000][0x104]!!
[2020/8/15 18:55:44-933] _lx_nand_flash_open5[13]
[2020/8/15 18:55:44-934] _lx_nand_flash_open7[ffffffff]
[2020/8/15 18:55:44-934] _lx_nand_flash_dev_extra_bytes_get->[6][1][4]
[2020/8/15 18:55:44-934] SPINAND_Read_Page_SpareData: [0xc0800]!!
[2020/8/15 18:55:44-934] _lx_nand_flash_dev_extra_bytes_get2->[ffffffff]
[2020/8/15 18:55:44-934] _lx_nand_flash_dev_page_erased_verify->[6][1]
[... 30 lines cut ...]
[2020/8/15 18:55:45-054] _lx_nand_flash_open5[13]
[2020/8/15 18:55:45-054] _lx_nand_flash_open7[ffffffff]
[2020/8/15 18:55:45-074] _lx_nand_flash_dev_extra_bytes_get->[9][1][4]
[2020/8/15 18:55:45-074] SPINAND_Read_Page_SpareData: [0x120800]!!
[2020/8/15 18:55:45-074] _lx_nand_flash_dev_extra_bytes_get2->[ffffffff]
[2020/8/15 18:55:45-074] _lx_nand_flash_dev_page_erased_verify->[9][1]
[2020/8/15 18:55:45-074] SPINAND_Read: [0x120800][0x800]!!
[2020/8/15 18:55:45-093] _lx_nand_flash_open15[0]

Here is my log, But nand_flash->nand_flash_max_mapped_sector = 0.

When Writing BootSector, There is LX_SECTOR_NOT_FOUND returned by __lx_nand_flash_logical_find.

2. LX_NAND_PAGE_EXTRA_INFO_STRUCT 
As your following,
typedef struct LX_NAND_EXTRA_INFO_STRUCT
{
      ULONG lx_nand_extra_info_sector;
}
  
3. SPI NAND Drivers.
I Have tested by RAW READ/WRITE, And no problem is found.

Best Regards
Joshua Wang

TiejunMS, 2020-08-21:
Hi @wxp821217 , did you erase nand flash and format file system in your application every time? If not, could you show us your application code related to levelx and filex only? And also, which levelx driver did you use?

wxp821217, 2020-08-21:
Dear Tiejun,

  I erased & format fs every time,  And the application is your demo_nand_flash_lx demo.

1.  The SPI NAND Driver is ported by myself.
2.  Some other log if you needed, I will test as your followings.
3.  My code is in-controlled, I Will ASk for my Boss'help.

Thanks.
Best Regards
JoshuaWang

TiejunMS, 2020-08-21:
The demo_nand_flash_lx runs on ram disk. So we have to erase all and format file system at the beginning. I guess this is where you get confused. On a real device, you only need to erase and format once.

closed 2020-08-25 by TiejunMS

wxp821217, 2020-08-27:
Dear Tiejun,

   If it is possible, Could you offer some contract info for this issue.

Best Regards
Joshua Wang

TiejunMS, 2020-08-27:
@wxp821217 , what do you mean by "contract info"?

goldscott, 2020-08-27:
I think that's supposed to be "contact info." Please contact AzureRTOSDev@microsoft.com

wxp821217, 2020-08-28:
@goldscott   💕👍