eclipse-threadx/levelx #6: LevelX doesnot work-well

wxp821217, 2020-09-03:
Dear Scott & Tiejun,
 
 I am running LevelX & FileX with your demo API, demo_flilx_nand_flash.c.
 
Platform -> ARMR5 Comminication Processor
NAND   -> SPI NAND winbond 1G.
Driver   -> SPI NAND Driver is ported by myself.
 
  The result looks like abnormal with LevelX.
 
[image]

 Here is detailed information with your demo.
1.     fx_media_format() is called firstly.
 
Because of no bad blocks, The routine is very simple.
Routine 1:
The Code Sample:
[image]

The log info:
[image]
 
Routine 2
The Code Sample:
[image]

The log info:
[image]
 
Routine3
The Code Sample:
[image]

The Log info:
[image]

 
2.     SECTOR_WRITE -> lx_nand_flash_sector_wite()
[image]

 
Routine1:
[image]

The Log Info:
[image]

 
Routine2:
[image]

The log info:
[image]

 
As you seen, The Write Operation is failed. BUT I donot know the reason.
 
Pls give me some handle.
Thanks very much.
 
Best Regards
Joshua Wang

xiuwencai, 2020-09-07:
Hi Joshua, from your log info, we can see that the _lx_nand_flash_physical_page_allocate has failed to allocate a new page. You may check the value of nand_flash -> lx_nand_flash_free_pages when calling _lx_nand_flash_physical_page_allocate and see if it is zero. You may also check if your driver can read and write the spare bytes correctly.

wxp821217, 2020-09-08:
Dear wencai,
 
   It returned here with error.    
[image]
 
  If the rouitne is checked by yourself as my words, There is only reading spare area method, So I think The problem is have nothing with SPI ANND Drivers.

  Thanks.

Best Regards
Joshua Wang

xiuwencai, 2020-09-09:
Hi Joshua, you may go through the lx_nand_flash_open function and see why lx_nand_flash_free_pages is zero. If lx_nand_flash_bad_blocks contains a large number, then check the driver_block_status_get function and verify if it returns correct block status. You may also check the values of lx_nand_flash_mapped_pages and lx_nand_flash_obsolete_pages. Checking variables lx_nand_flash_diagnostic_* in the LX_NAND_FLASH struct may also help you diagnosing the problem.

wxp821217, 2020-09-09:
Dear wencai,
  
    There is no bad block. As you seen, All return calue of river_block_status_get  is 0xFF. 
[image]

Best Regards
Joshua Wang

xiuwencai, 2020-09-09:
Can you post all the variable values inside the nand_flash struct?

wxp821217, 2020-09-10:
Dear wencai,

   * @ Ending of LX_NAND_FLASH_OPEN API  
OR 
   * @ Failure of WRITE BOOT Sector..

   Pls help to confirm the scene, I will post all the information you mentioned.
Best Regards
Joshua Wang

xiuwencai, 2020-09-11:
Please post the variable states at the end of _lx_nand_flash_open function.

wxp821217, 2020-09-11:
Dear wencai,

  Here is log info.
  
[2020/9/11 15:43:23-291]  nand_flash ->lx_nand_flash_state=0x[4E4F524F]!
[2020/9/11 15:43:23-291]  nand_flash ->lx_nand_flash_total_blocks=0x[A]!
[2020/9/11 15:43:23-291]  nand_flash ->lx_nand_flash_pages_per_block=0x[40]!
[2020/9/11 15:43:23-307]  nand_flash ->lx_nand_flash_bytes_per_page=0x[800]!
[2020/9/11 15:43:23-307]  nand_flash ->lx_nand_flash_words_per_block=0x[8000]!
[2020/9/11 15:43:23-307]  nand_flash ->lx_nand_flash_words_per_page=0x[200]!
[2020/9/11 15:43:23-323]  nand_flash ->lx_nand_flash_total_pages=0x[280]!
[2020/9/11 15:43:23-323]  nand_flash ->lx_nand_flash_bad_blocks=0x[0]!
[2020/9/11 15:43:23-323]  nand_flash ->lx_nand_flash_free_pages=0x[271]!
[2020/9/11 15:43:23-323]  nand_flash ->lx_nand_flash_mapped_pages=0x[1]!
[2020/9/11 15:43:23-340]  nand_flash ->lx_nand_flash_obsolete_pages=0x[4]!
[2020/9/11 15:43:23-340]  nand_flash ->lx_nand_flash_minimum_erase_count=0x[200A0]!
[2020/9/11 15:43:23-340]  nand_flash ->lx_nand_flash_maximum_erase_count=0x[54706191]!
[2020/9/11 15:43:23-356]  nand_flash ->lx_nand_flash_free_block_search=0x[0]!
[2020/9/11 15:43:23-356]  nand_flash ->lx_nand_flash_found_block_search=0x[0]!
[2020/9/11 15:43:23-357]  nand_flash ->lx_nand_flash_found_page_search=0x[1]!
[2020/9/11 15:43:23-374]  nand_flash ->lx_nand_flash_max_mapped_sector=0x[14706191]!
[2020/9/11 15:43:23-375]  nand_flash ->lx_nand_flash_page_corrections=0x[0]!
[2020/9/11 15:43:23-375]  nand_flash ->lx_nand_flash_last_block_correction=0x[0]!
[2020/9/11 15:43:23-375]  nand_flash ->lx_nand_flash_last_page_correction=0x[0]!
[2020/9/11 15:43:23-391]  nand_flash ->lx_nand_flash_diagnostic_system_errors=0x[9]!
[2020/9/11 15:43:23-391]  nand_flash ->lx_nand_flash_diagnostic_system_error=0x[5B]!
[2020/9/11 15:43:23-391]  nand_flash ->lx_nand_flash_diagnostic_sector_write_requests=0x[0]!
[2020/9/11 15:43:23-407]  nand_flash ->lx_nand_flash_diagnostic_sector_read_requests=0x[0]!
[2020/9/11 15:43:23-407]  nand_flash ->lx_nand_flash_diagnostic_sector_release_requests=0x[0]!
[2020/9/11 15:43:23-422]  nand_flash ->lx_nand_flash_diagnostic_page_allocates=0x[0]!
[2020/9/11 15:43:23-422]  nand_flash ->lx_nand_flash_diagnostic_page_allocate_errors=0x[0]!
[2020/9/11 15:43:23-422]  nand_flash ->lx_nand_flash_diagnostic_sector_mapping_cache_hits=0x[0]!
[2020/9/11 15:43:23-438]  nand_flash ->lx_nand_flash_diagnostic_sector_mapping_cache_misses=0x[0]!
[2020/9/11 15:43:23-438]  nand_flash ->lx_nand_flash_diagnostic_sector_mapping_cache_hits=0x[0]!
[2020/9/11 15:43:23-454]  nand_flash ->lx_nand_flash_diagnostic_sector_mapping_cache_misses=0x[0]!
[2020/9/11 15:43:23-454]  nand_flash ->lx_nand_flash_diagnostic_page_extra_bytes_cache_hits=0x[B]!
[2020/9/11 15:43:23-472]  nand_flash ->lx_nand_flash_diagnostic_page_extra_bytes_cache_misses=0x[5]!
[2020/9/11 15:43:23-472]  nand_flash ->lx_nand_flash_diagnostic_page_0_cache_hits=0x[E]!
[2020/9/11 15:43:23-472]  nand_flash ->lx_nand_flash_diagnostic_page_0_cache_misses=0x[C]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_block_status_cache_hits=0x[14]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_block_status_cache_misses=0x[A]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_block_reclaim_attempts=0x[0]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_block_erases=0x[2]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_block_status_gets=0x[A]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_block_status_sets=0x[0]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_page_extra_bytes_sets=0x[4]!
[2020/9/11 15:43:23-515]  nand_flash ->lx_nand_flash_diagnostic_page_writes=0x[A]!
[2020/9/11 15:43:23-530]  nand_flash ->lx_nand_flash_diagnostic_block_status_cache_misses=0x[A]!
[2020/9/11 15:43:23-530]  nand_flash ->lx_nand_flash_diagnostic_page_extra_bytes_gets=0x[5]!
[2020/9/11 15:43:23-546]  nand_flash ->lx_nand_flash_diagnostic_page_reads=0x[C]!
[2020/9/11 15:43:23-546]  nand_flash ->lx_nand_flash_diagnostic_moved_pages=0x[0]!
[2020/9/11 15:43:23-546]  nand_flash ->lx_nand_flash_diagnostic_block_erased_verifies=0x[5]!
[2020/9/11 15:43:23-563]  nand_flash ->lx_nand_flash_diagnostic_page_erased_verifies=0x[5]!
[2020/9/11 15:43:23-563]  nand_flash ->lx_nand_flash_diagnostic_initial_format=0x[0]!
[2020/9/11 15:43:23-563]  nand_flash ->lx_nand_flash_diagnostic_erased_block=0x[5]!
[2020/9/11 15:43:23-579]  nand_flash ->lx_nand_flash_diagnostic_re_erase_block=0x[2]!
[2020/9/11 15:43:23-579]  nand_flash ->lx_nand_flash_diagnostic_page_being_obsoleted=0x[1]!
[2020/9/11 15:43:23-598]  nand_flash ->lx_nand_flash_diagnostic_page_obsoleted=0x[1]!
[2020/9/11 15:43:23-598]  nand_flash ->lx_nand_flash_diagnostic_page_obsoleted=0x[0]!
[2020/9/11 15:43:23-598]  nand_flash ->lx_nand_flash_diagnostic_mapping_invalid=0x[8]!
[2020/9/11 15:43:23-598]  nand_flash ->lx_nand_flash_diagnostic_mapping_write_interrupted=0x[0]!
[2020/9/11 15:43:23-638]  nand_flash ->lx_nand_flash_diagnostic_page_not_free=0x[0]!
[2020/9/11 15:43:23-638]  nand_flash ->lx_nand_flash_diagnostic_page_data_not_free=0x[4]!

Best Regards
Joshua Wang

xiuwencai, 2020-09-14:
From your log, I see some abnormal values. For example, the lx_nand_flash_minimum_erase_count and lx_nand_flash_maximum_erase_count are quite large and the difference between min and max is large. It indicates that there may be issues in reading/writing the pages. 
The lx_nand_flash_max_mapped_sector is quite large too. This indicates that invalid data may be returned by _extra_bytes_get. Please check your driver to see if it can access spare data correctly and make sure the extra data is stored in "user data" area of the spare area.
The non-zero values in lx_nand_flash_diagnostic_system_error(s), lx_nand_flash_diagnostic_erased_block, lx_nand_flash_diagnostic_re_erase_block and lx_nand_flash_diagnostic_mapping_invalid all show that there may be some data corruption in reading/writing the flash. My suggestion is to take the simulator as reference and check your driver and hardware, and make sure the written data can be read out correctly.

wxp821217, 2020-09-16:
Dear wencai,

   I will check it.
   if some demo with real nand flash is ok, Pls  show us.
  Thanks.

Best Regards
Joshua Wang

yuxin-azrtos, 2020-09-25:
@wxp821217 we do not have demos using NAND flash available.

goldscott, 2020-09-25:
Hi @wxp821217 - any updates? Have you made any progress or have any questions?

goldscott, 2020-09-29:
I'll close this ticket for now but feel free to re-open.

closed 2020-09-29 by goldscott