eclipse-threadx/levelx #9: Questions about the use of ECC

aod-ad, 2021-04-09:
Hello, I found some questions about the use of ECC:
       In the block, it should be ensured that each page is written only once, and then the ECC data of the page can be written into the OOB. However, since the erasing times of the block are stored in the 0th page, it will result in the 0th page Data will be written multiple times, resulting in unequal ECC data each time, which will cause ECC error correction errors.
        On line 286 and line 296 in lx_nand_flash_open.c

yuxin-azrtos, 2021-04-14:
Hi @aod-ad, thank you for reporting the bug.  We will be working on a fix.

bhnaphade, 2021-05-19:
We are investigating the issue and will update this issue when the code is fixed

mrdrinnan, 2021-07-13:
I am also having this problem.

mrdrinnan, 2021-07-13:
The originally mentioned problem is present in the code in several locations, which can be readily found by searching for instances of LX_BLOCK_ERASED.
After implementing a workaround for the original problem (only do one write instead of two), another problem was found.

It occurs in lx_nand_flash_block_full_update.c, where a write to page 0 is performed when the block is full. ECC fails in an identical manner to the originally described problem: you can't write a page without first erasing, and writing LX_NAND_PAGE_LIST_VALID to page 0 is a second page write!

trinhnhatnam, 2021-08-30:
It doesn't seem LevelX cares about "programs per page" limit. I stepped through the code and saw that page 0's are repeatedly programmed without the block getting erased.

For example, at format time, block 0 page 0 gets programmed 5 times. Most SLC allows 4 programs per page, so 5 violates the limit but most probably can still work. However, some NANDs only allow one program per page, so 5 will be a problem as I found out.

Please prove me wrong, but I'm losing faith in AzureRTOS, at least with LevelX and NAND.

zbqxyz, 2021-11-01:
@yuxin-azrtos the 6.19  is fixed?

Christian-Sander, 2022-09-16:
Any news on this? It seems that ECC in NAND flashes isn't functional at all right now. I'm also unsure if the extra data is meant to be protected by ECC in any way. NAND flashes such as the Winbond SLC chips also use separate ECC for the extra data when hardware ECC is used. I think this would required either writing tthis data only once, erasing the block before doing that or maybe writing in different places of the spare region with multiple ECCs if there is enough space?

 I also haven't seen any public implementation of FileX+LevelX with a NAND flash that implements ECC. There is https://github.com/STMicroelectronics/x-cube-azrtos-h7/blob/main/Projects/STM32H743I-EVAL/Applications/FileX/Fx_Nand_Write_Read_File/LevelX/Target/lx_stm32_nand_custom_driver.c but that is just a basic implementation without ECC as far as I can see.

Are there any plans to fix the existing problems? If so, when can a working implementation be expected?
The last comments I've seen were from February.

I'm also inclined to believe that I should look for other solutions (maybe LittleFS for file system + wear levelling or dhara in combination with FileX?).
I would gladly be proven wrong as I have good experience with the ThreadX/NetX parts.

bhnaphade, 2022-09-16:
@Christian-Sander : We are working on level nand fix to resolve the ECC issue and are currently in process of testing the new updates. The fix should be available soon.

ReniferDev, 2023-05-12:
@bhnaphade Hello. I've seen the release of 6.2.1 with description "Fixed NAND ECC issues by new NAND logic". Is this issue solved by now? Is there any mechanism that limits the number of page rewrites (before mandatory block erase) by LevelX? Do You have any example of ECC use?

xiuwencai, 2023-05-25:
@ReniferDev The new NAND logic will not rewrite pages, so ECC can be used now. Samples will be added later.

closed 2023-06-13 by bhnaphade