eclipse-threadx/levelx #10: ECC calibration problem

SmartSmallBoy, 2021-05-18:
Hello! In the process of using it, I found that page 0 will be written many times in “lx_nand_flash_open.c ". On lines 286 and 296, I can only block the ECC verification on page 0 before I can use it normally.

bhnaphade, 2021-05-18:
Hi @SmartSmallBoy,  We will be working on a fix.

closed 2021-05-20 by SmartSmallBoy

zbqxyz, 2021-08-23:
> 
> 
> Hello! In the process of using it, I found that page 0 will be written many times in “lx_nand_flash_open.c ". On lines 286 and 296, I can only block the ECC verification on page 0 before I can use it normally.

Is this problem solved?

zbqxyz, 2021-08-23:
> 
> 
> Hi @SmartSmallBoy, We will be working on a fix.

hi @bhnaphade Is this problem solved?

SmartSmallBoy, 2021-08-23:
Hi！It seems not. I read the code of this function in version 6.18 and there is no change

bhnaphade, 2021-08-23:
We are currently working on the issue