eclipse-threadx/levelx #16: Address alignment issue while writing to internal falsh of stm32u5

Nelthrion, 2022-06-20:
HI

I am using the STM32U575 for my project with the internal nor flash provided.
Noticing that the address are word(32bits) aligned while writing to the flash, but for STM32U5 internal flash the address needs to be quadword(128bits) aligned. Is there a configuration in levelx (like a line size) with which levelx will provide me quadword aligned addresses, or I need to take care of this specifically in the flash write driver routine(I think will affect the performance of write operation)?

Thanks

xiuwencai, 2022-06-23:
Hi Nelthrion, 
Unfortunately, LevelX does not support this internal NOR flash. There is no such configuration provided for now. We will add support for STM32 internal NOR flash in future releases.

Nelthrion, 2022-06-27:
Thanks for the reply. I will continue to use external flash for now.