eclipse-threadx/levelx #17: Substraction underflow in lx_nor_flash_open.c

GiHuCo, 2022-07-13:
Hi,
I am using Filex + LevelX (standalone). But every try I received LX_ERROR. When I investigated the occuring error, I came to that point in lx_nor_flash_open where
`used_sectors =  sectors_per_block - free_sectors;`
is assigned. In my case free_sectors was of a greater value than sectors_per_block (128 > 126) and thereby the substraction lead to an underflow and the following code execution failed.
I am assuming that decrementing 1 when assingning the value to sectors_per_block causes the issue:
`sectors_per_block =  (nor_flash -> lx_nor_flash_words_per_block / LX_NOR_SECTOR_SIZE) - 1;`

xiuwencai, 2022-07-14:
Hi @GiHuCo, it seems the free bit map was corrupted. Have you tried erasing all the blocks to reinitialize the format? If the issue still exists, please check the flash driver to make sure it is accessing the flash correctly.

GiHuCo, 2022-07-15:
Hi,
thanks for your help. The important thing which lead us into that trouble and and lots of time consuming debugging: **The custum nor flash must be initialised with all ones instead of all zeros!!**

closed 2022-07-26 by bhnaphade