eclipse-threadx/levelx #20: _lx_nor_flash_logical_sector_find costs a lot of time, making LevelX writes quite slow

HaoboGu, 2023-08-03:
I'm using Filex with LevelX nor flash driver. I found that when writing a lot of data, _lx_nor_flash_logical_sector_find costs a lot of time(about 50ms each write), making LevelX write quite slow (about 5kb/s).

I checked the source code of LevelX, this while block costs the most of the time. Is there anyway to improve the write speed of LevelX?
```c
while (total_blocks--) 
    {

        /* Setup the block word pointer to the first word of the search block.  */
        block_word_ptr =  (nor_flash -> lx_nor_flash_base_address + (i * nor_flash -> lx_nor_flash_words_per_block));

        /* Determine if the minimum and maximum logical sector values are present in the block header.  If these are 
           present, we can quickly skip blocks that don't have our sector.  */

        /* Read the minimum and maximum logical sector values in this block.  */
#ifdef LX_DIRECT_READ
        
        /* Read the word directly.  */
        min_logical_sector =  *(block_word_ptr + LX_NOR_FLASH_MIN_LOGICAL_SECTOR_OFFSET);
#else
        status =  _lx_nor_flash_driver_read(nor_flash, block_word_ptr + LX_NOR_FLASH_MIN_LOGICAL_SECTOR_OFFSET, &min_logical_sector, 1);

        /* Check for an error from flash driver. Drivers should never return an error..  */
        if (status)
        {
        
            /* Call system error handler.  */
            _lx_nor_flash_system_error(nor_flash, status);

            /* Return the error.  */
            return(status);
        }
#endif
#ifdef LX_DIRECT_READ
        
        /* Read the word directly.  */
        max_logical_sector =  *(block_word_ptr + LX_NOR_FLASH_MAX_LOGICAL_SECTOR_OFFSET);
#else
        status =  _lx_nor_flash_driver_read(nor_flash, block_word_ptr + LX_NOR_FLASH_MAX_LOGICAL_SECTOR_OFFSET, &max_logical_sector, 1);

        /* Check for an error from flash driver. Drivers should never return an error..  */
        if (status)
        {
        
            /* Call system error handler.  */
            _lx_nor_flash_system_error(nor_flash, status);

            /* Return the error.  */
            return(status);
        }
#endif

        /* Are the values valid?  */
        if ((min_logical_sector != LX_ALL_ONES) && (max_logical_sector != LX_ALL_ONES))
        {

            /* Now let's check to see if the search sector is within this range.  */
            if ((logical_sector < min_logical_sector) || (logical_sector > max_logical_sector))
            {

                /* Move to the next block.  */
                i++;
      
                /* Determine if we have wrapped.  */
                if (i >= nor_flash -> lx_nor_flash_total_blocks)
                {
        
                    /* Yes, we have wrapped, set to block 0.  */
                    i =  0;
                }

[... 164 lines cut ...]
            
                /* Yes, wrap around, go back to the beginning.  */
                j =  0;
            }
        }

        /* Determine if there are any more mapped sectors.  */
        if (mapped_sectors == 0)
            break;
            
        /* Move to the next block.  */
        i++;
       
        /* Determine if we have wrapped.  */
        if (i >= nor_flash -> lx_nor_flash_total_blocks)
        {
        
            /* Yes, we have wrapped, set to block 0.  */
            i =  0;
        }

        /* Start at the first sector in the next block.  */
        j =  0;
    }
```

HaoboGu, 2023-12-01:
@xiuwencai thanks, I'll try it

HaoboGu, 2023-12-07:
@xiuwencai a quick question: when should I call `lx_nor_flash_extended_cache_enable`? before or after `lx_nor_flash_open`?

xiuwencai, 2023-12-22:
Sorry for late reply. Call lx_nor_flash_extended_cache_enable after lx_nor_flash_open.

DNTT-1603, 2025-07-07:
@HaoboGu , hi! Did you test with `lx_nor_flash_extended_cache_enable`? Does it actually improve write time?