eclipse-threadx/levelx #34: LevelX initialization time problem

SimonTechv, 2024-02-10:
I ported levelX to work with the STM32F412ZGT6 board and the MICRON N25Q128 SPI flash memory. The volume used is 4096 sectors of 4 kilobytes each.

Before starting to work with flash memory, I call the function:

`
_lx_nor_flash_open(&nor_mem_desc, "VOL0:", flash_driver_init);
`

Its execution takes about 1 second. In the image below, I provide the necessary information:

[image]

I looked through the code and found that during the initialization process levelX reads all blocks of flash memory. 

Is it possible to reduce the startup time without reducing the memory size? 
I plan to use 1 Gbit NOR memory, but in this case the initialization time will increase several times more. 

For more information, I have uploaded the [project code](https://github.com/SimonTechv/azure_stm32_ftl)

rahmanih, 2024-02-14:
Hi @SimonTechv,
Could you try to customize the "lx_user.h" by enabling the 2 following flags:
* https://github.com/eclipse-threadx/levelx/blob/ec683ddf1461a05e462d301bf8a15124136b2ca0/common/inc/lx_user_sample.h#L97
* https://github.com/eclipse-threadx/levelx/blob/ec683ddf1461a05e462d301bf8a15124136b2ca0/common/inc/lx_user_sample.h#L106

after enabling this flags you'll need to call in the API: [lx_nor_flash_extended_cache_enable](https://github.com/eclipse-threadx/levelx/blob/ec683ddf1461a05e462d301bf8a15124136b2ca0/common/inc/lx_api.h#L772)
just after the <code>lx_nor_flash_open() </code>.

SimonTechv, 2024-02-16:
@rahmanih, I tried to enable all the options you described, but it did not affect the initialization speed.

Sometimes an INVALID_BLOCK error occurs during operation, which never disappears. Is this normal behavior or does it require careful investigation?

rahmanih, 2024-02-19:
> @rahmanih, I tried to enable all the options you described, but it did not affect the initialization speed.
> 
> Sometimes an INVALID_BLOCK error occurs during operation, which never disappears. Is this normal behavior or does it require careful investigation?
> 
> UPD: I found critical error in QSPI NOR DRIVER in Page programm algorithm. I will fix it and retest module soon.

could you try to test with 64K block size?

regards
haithem.

rahmanih, 2024-02-21:
@SimonTechv 
> I changed the sector size to 64 KB. Initialization time has been reduced to 700 milliseconds. However, when I did fault tolerance testing, I found out that a sudden reset damages the internal data structure. In my case, after 10 random resets while recording, I get an error. The error appears [here](https://github.com/eclipse-threadx/levelx/blob/master/common%2Fsrc%2Flx_nor_flash_open.c#L409).
after switching to 64KB, you'll need to erase the NOR flash and reformat it, did you do that?

SimonTechv, 2024-02-21:
@rahmanih Yes, I have completed these operations.

rahmanih, 2024-02-21:
>I found out that a sudden reset damages the internal data structure
>For more information, I have uploaded the [project code](https://github.com/SimonTechv/azure_stm32_ftl)
Sorry but there isn't any 'fault tolerance' code within your project.

SimonTechv, 2024-02-21:
@rahmanih Protection against power failures is littered in the LevelX documentation. I'm using a random reset generator for the microcontroller, so my code just uses sequential writing.

rahmanih, 2024-02-21:
> @rahmanih Protection against power failures is littered in the LevelX documentation. I'm using a random external reset generator for the microcontroller, so my code just uses sequential writing.

unless I'm missing something, but I think the fault tolerance is managed at FileX.
https://github.com/eclipse-threadx/rtos-docs/blob/main/rtos-docs/filex/chapter6.md

SimonTechv, 2024-02-21:
@rahmanih Official Docs page
[image]

SimonTechv, 2024-02-21:
I noticed this line. There are many more 512 byte blocks in a 64 kilobyte block than described in the [comment](https://github.com/eclipse-threadx/levelx/blob/master/common%2Fsrc%2Flx_nor_flash_simulator.c#L33)

rahmanih, 2024-05-15:
closing this issue as solution is provided.

closed 2024-05-15 by rahmanih