eclipse-threadx/levelx #51: lx_nand_flash_sector_release.c put LX_NULL into main buffer pointer when call lx_nand_flash_driver_pages_read function

TheBraveBurjak, 2024-12-27:
lx_nand_flash_sector_release.c
```
                /* Read a page.  */
#ifdef LX_NAND_ENABLE_CONTROL_BLOCK_FOR_DRIVER_INTERFACE
                status = (nand_flash -> lx_nand_flash_driver_pages_read)(nand_flash, block, (ULONG)page, LX_NULL, spare_buffer_ptr, 1);
#else
                status = (nand_flash -> lx_nand_flash_driver_pages_read)(block, (ULONG)page, LX_NULL, spare_buffer_ptr, 1);
#endif
```
Hello, this LX_NULL is crashing my application because I didn't expect LevelX gonna use this function to read spare bytes. Especially, when we have defined extra_bytes_get. Maybe we should use extra_bytes_get instead?

```nand_flash_ptr -> lx_nand_flash_driver_extra_bytes_get =        lx_nand_driver_extra_bytes_get;```

fdesbiens, 2025-01-15:
Thank you for the bug report @TheBraveBurjak. Can someone in @eclipse-threadx/iot-threadx-committers take a look?

Chabrol, 2025-01-20:
But [NAND Driver Requirements](https://github.com/eclipse-threadx/rtos-docs/blob/main/rtos-docs/levelx/chapter3.md#nand-driver-requirements) doesn't state that lx_nand_flash_driver_extra_bytes_get must be implemented. I would rather keep it the original way and instead improve the [Driver Read Pages](https://github.com/eclipse-threadx/rtos-docs/blob/main/rtos-docs/levelx/chapter3.md#driver-read-pages) description, mentioning possible null arguments.